QuentinTournier40/AnimationFreeCAD #14: clapCinema.svg not detected with FreeCAD appImage

RockVicedomini, 2023-02-13:
Hello,
I just installed AnimationFreeCAD wb through the Addon Manager. 
The workbench seems fully functional, but FreeCAD is continuously reporting an error every time I open a new file:

`File does not exist! 
/tmp/.mount_FreeCANocsaG/usr/Mod/AnimationFreeCAD/icons/clapCinema.svg`  

I am using the latest FreeCAD 0.21 weekly build for linux on KDE Neon OS, but the error is present also with the stable release 0.20.2

[code]
OS: KDE neon 5.26 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.31850 (Git) AppImage
Build type: Release
Branch: master
Hash: d8d3bb8120ad139cf1eb2acf0be0ae6c297ac7f3
Python 3.10.9, Qt 5.15.6, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Beltrami 1.0.8-alpha
  * Nodes 0.1.36
  * freecad.gears 1.0.0
  * FEMbyGEN 1.2.0
  * Render 2022.4.0
  * Assembly4 0.12.5
  * Movie 2022.12.20
  * Assembly3 0.11.4
  * Curves 0.6.6
  * MeshRemodel 1.8919.0
  * A2plus 0.4.60i
  * freecad-dynamics-workbench-master
  * CfdOF 1.21.4
  * Glass
  * lattice2 1.0.0
  * AnimationFreeCAD 1.0.0
  * fasteners 0.4.54
  * Plot 2022.4.17
  * ExplodedAssembly
[/code]

gusantor, 2023-02-16:
Hello

On Ubuntu 20.04.5 LTS
 
I'm using 

./FreeCAD_0.20.2-2022-12-27-conda-Linux-x86_64-py310.AppImage

Animation workbench version 1.0.0

And getting similar message

Cannot find icon: /tmp/.mount_FreeCAUBDS7u/usr/share/../Mod/AnimationFreeCAD/icons/clapCinema.svg

Hopefully it helps some way.

Thanks

RDSH-KS, 2023-02-18:
Similar with windows 11.   Error Message: C:/Program Files/FreeCAD 0.20/Mod/AnimationFreeCAD/icons/clapCinema.svg. 
Actually, there is not folder named AanimationFreeCAD in the path .../Mod/... .  
the clapcinema.svg file is found in the folder C:\Users\xxx\AppData\Romaing\FreeCAD\Mod|AnimationFreeCAD\icons\.

OtacilioNeto, 2023-03-27:
Same problem here using Ubuntu 22.04

```
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32532 (Git) AppImage
Build type: Release
Branch: master
Hash: 20e44eba50e1365b9e7bb9c5c244afe96a627a28
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Portuguese/Brazil (pt_BR)
Installed mods: 
  * MeshRemodel 1.8919.0
  * Manipulator 1.5.0
  * ExplodedAssembly
  * fasteners 0.4.55
  * Assembly4 0.12.6
  * kicadStepUpMod 10.17.5
  * dodo 1.0.0
  * AnimationFreeCAD 1.0.0
  * 3DfindIT 1.2.0

```

RockVicedomini, 2023-03-30:
I was able to solve the issue by merging the following patch proposed by @chennes 

https://github.com/QuentinTournier40/AnimationFreeCAD/pull/11/commits/befe8ec593067094700ba55e2dc01ed211861737

ThomasHarder, 2023-09-09:
Thanks @chennes. Same problem fixed!!!

closed 2025-03-03 by QuentinTournier40