QuentinTournier40/AnimationFreeCAD #18: When we are using a dark theme in FreeCAD, the Animation WB's icon cannot be seen.

raindropsfromsky, 2025-03-02:
When we use a dark theme in FC, the icon for the Animation WB button looks blank.

Normal mode: [image]
with mouse hover: [image]

**Desired:** 
Please use a color-scheme that is usable in dark theme also. 
(Now dark themes are popular in FC.)

raindropsfromsky, 2025-03-03:
On closer inspection, I found that this is because the icon image is not loaded:

On start, FC shows this error:
`Cannot find icon: C:\Program Files\Util\FreeCAD\data\../Mod\AnimationFreeCAD\icons\clapCinema.svg
`
The log file shows this warning:
`Wrn: Cannot find icon: C:\Program Files\Util\FreeCAD\data\../Mod\AnimationFreeCAD\icons\clapCinema.svg`

QuentinTournier40, 2025-03-03:
Hey,

Could you retry the install now please

raindropsfromsky, 2025-03-03:
Yes, now it is visible in normal mode and when my mouse hovers on it. 
[image]

Thanks!

closed 2025-03-03 by raindropsfromsky