QuentinTournier40/AnimationFreeCAD #20: Unable to download the help documentation for the Animation WB

raindropsfromsky, 2025-03-06:
I am using the Animation WB for the very first time, and I am trying to read the help docs.

I reached this [help page](https://github.com/QuentinTournier40/AnimationFreeCAD/commit/9af827da907676d429137e8fc23259b9898d5525#diff-71860b07bf846e3e3c7491d6da043bfb37b53bfecba375de7a4713356c7cce5c), but there is no text there.

I had to search hard to locate the [user guide](https://github.com/QuentinTournier40/AnimationFreeCAD/blob/9af827da907676d429137e8fc23259b9898d5525/docs/guide/en/user's%20guide/User's%20guide.pdf) pdf file, as it is hidden inside other code. 

**Desired:**
1. Please make it easy for non-coders to reach the help file. There should be a direct download link at the top.
2. As a non-coder, I have no clue how to make use of the init and py files. There should be enough guidance for this.

QuentinTournier40, 2025-03-06:
Here is the link to download the English user guide : [doc](https://github.com/QuentinTournier40/AnimationFreeCAD/blob/main/docs/guide/en/user's%20guide/User's%20guide.pdf).

I hope it will be helpful. However, we wrote this document at the request of the university 3 years ago, and we only translated it from French, so there may be some translation errors.

But I don't understand why you would like to know "how to make use of the init and py files.". Do you mean there should be documentation about the code written so other people could participate in the project ?

But I don't understand why you would like to know "how to make use of the init and py files.". Are you suggesting that there should be documentation about the code so that others can participate in the project?

raindropsfromsky, 2025-03-06:
> Here is the link to download the English user guide : [doc](https://github.com/QuentinTournier40/AnimationFreeCAD/blob/main/docs/guide/en/user's%20guide/User's%20guide.pdf).

Thanks, but the users won't be looking at GitHub issues to discover this link.
Therefore, it is best to provide this direct link in the Addon Manager's page for the Animation WB.
[image]

> I don't understand why you would like to know "how to make use of the init and py files.". Do you mean there should be documentation about the code written so other people could participate in the project ?

I am not sure what these files do; especially because this code is listed in the "documentation" page.
(Usually, the code is not posted as documents.)

In many apps, a script is provided that the user has to copy and paste as a local file, and then edit it.
Finally he has to place this modified file in a certain installation folder, so that the app makes use of it.

It looked similar to that type of application. 
(As mentioned, I am not a coder, so I cannot guess the purpose of such files.)